Nature40/pimod #1: Allow stdin redirection (somehow?)

jonashoechst, 2019-01-23:
I tried to set a password in the chroot using `RUN bash -c '(echo natur; echo natur) | passwd pi'`, which failed due to the pipe I used. 

There could be a method (some fancy combination of escapes) to have the whole line executed inside the chroot, not only the first part. Please find out and document.

gh0st42, 2019-01-23:
`RUN bash -c 'echo -e "bushberry\nbushberry" | passwd pi'` works fine

gh0st42, 2019-01-23:
The errors just don't appear... 
`bash -c` doesn't work at all.

`RUN apt-get -y dist-upgrade` executes.
`RUN /bin/bash -c "apt-get -y dist-upgrade"` doesn't:
```
apt 1.4.8 (armhf)
Usage: apt-get [options] command
       apt-get [options] install|remove pkg1 [pkg2 ...]
       apt-get [options] source pkg1 [pkg2 ...]

apt-get is a command line interface for retrieval of packages
and information about them from authenticated sources and
for installation, upgrade and removal of packages together
with their dependencies.
[..]
```

geistesk, 2019-01-23:
I will take a look at this soon.

A hacky workaround would be creating an install script, `INSTALL`ing it, `RUN` it and `RUN rm` it afterwards.

commit d9723756 (2019-01-24): added bash -c wrapper, closes #1 #2
closed 2019-01-24 by jonashoechst