Nature40/pimod #2: Add PASSWD command

jonashoechst, 2019-01-23:
The default password of the pi user might be changed in many cases for custom distros, thus I suggest a PASSWD command. 

For now it can be done with: 

```
(echo thepassword; echo thepassword) | RUN passwd pi
```

gh0st42, 2019-01-23:
Instead of adding special commands I'd suggest having `RUN` execute its parameters in a `bash -c` block so that one can set environment variables and use pipes as expected.

geistesk, 2019-01-23:
As mentioned in #1, a hacky workaround would be creating an install script, INSTALLing it, RUN it and RUN rm it afterwards.

I will investigate this asap.

gh0st42, 2019-01-23:
https://github.com/Nature40/pi-mod/blob/ab3785662fc96511c134429f28053b8fec0699e3/30-chroot.sh#L49

Wrap the `$@` with quotes and `RUN bash -c "bla" works``

`proot -0 -q qemu-arm-static -w / -r $CHROOT_MOUNT "$@"`

Didn't test for side effects. Still bash -c should be part of the RUN code

jonashoechst, 2019-01-23:
> Instead of adding special commands I'd suggest having `RUN` execute its parameters in a `bash -c` block so that one can set environment variables and use pipes as expected.

Piping to passwd is already kind of hackish. As stated before setting a default password is a common thing for a distro and thus would be a good idea to be implemented as an own function.

gh0st42, 2019-01-23:
Actually that IS the unix way.. and it used like this and in similar ways across various linux and unix systems for setting passwords. Wrapping the same code in a helper doesn't make it better :) If you want a wrapper you could use `raspi-config` in headless mode to change the pass for you.

Putting `RUN` in the wrong places oder adding `bash -c` on the other hand IS hackish :D

adur1990, 2019-01-24:
Why arguing about the "standard" unix way? If we want it the standard unix way, why having a RUN command at all? Why having this kind of configuration? Just write bash scripts. This is how unix works.

So, I don't get you point, @gh0st42

gh0st42, 2019-01-24:
Actually the `Pifile` is just a bash script :P And that is not the core of the unix philosophy.
The point is that hiding stuff like this in non-generic commands increases the time needed to maintain the project as the command is likely to break over time. Also it might not be as distro agnostic as one might think (e.g. some implementations of passwd need additional parameters to accept a password from stdin). Therefore, if one expects PASSWD to just work and it fails (and currently doesn't output an error), it is much harder to debug or alter for another base distro. Having only the minimal required toolset in the project makes it easier to maintain it in the long run. Stuff like setting up network, external interfaces or changing passwords often change over time within distros. 
This project could easily be useful for many different users and distros. Having a clear scope and focus on rock stable core functionality that can easily be extended by other projects is a big selling point.

jonashoechst, 2019-01-24:
your suggestion is good, but won't work, since a Pifile still is a bash file and thus every pipe (not escaped or in quotes) will be interpreted before handed to `bash -c`. Thus the best we can get is:

```
RUN 'hexdump /dev/urandom | head'
```
or 
```
RUN hexdump /dev/urandom \| head
```

geistesk, 2019-01-24:
Your first line was my suggestion - giving a single string as the only parameter to the `RUN` function.

jonashoechst, 2019-01-24:
won't break the functionality as used right now, right? If it doesn't, the suggestion is okay for me

gh0st42, 2019-01-24:
what about the approach used in pi-gen? they mostly use heredoc's:
https://github.com/RPi-Distro/pi-gen/blob/d945d422d4e26df6fff7185da28874432a974d78/stage2/01-sys-tweaks/01-run.sh#L14

implemented here:
https://github.com/RPi-Distro/pi-gen/blob/4fc1423fdff262f1ca8e5f4a0ba706b5ff35f428/scripts/common#L74

the current solution could be used like this, couldn't it?

geistesk, 2019-01-24:
Related to the closing commit, am I mistaking or do I have to surround my `RUN` command now with quotation signs?

```
RUN apt-get update
# Fails and just shows the apt-get help page

RUN 'apt-get update'
# Works
```

gh0st42, 2019-01-24:
Yes, also had this problem yesterday when playing with fixes. That's why I didn't suggest this with my quotes fix. It didn't feel completely fixed or right.

gh0st42, 2019-01-24:
Since bash -c in RUN only executes the first command all Pifiles with `RUN bash -c whatever` just execute the bash and drop you on the command prompt.

https://github.com/Nature40/pimod/commit/c50773d6296011299133d7b5edbdc15680a8584d at least made it possible to use here documents very comfortable, including pipes and stuff.

```
RUN << EOF
hexdump /dev/urandom | head
ifconfig
EOF
```

and if needed one could write `RUN bash -c ...` eventhough grouping command together makes more sense due to the overhead of mounting and prooting.

commit 2a28078a (2019-01-24): added bash -c wrapper, closes #2