Nature40/pimod #3: Investigate failing dist-upgrade

jonashoechst, 2019-01-23:
A dist-upgrade fails for me, I guess in consequence to the simulated system. Please investigate this behavior and propose a workaround / solution if possible.

```
Reading package lists... Done
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Calculating upgrade... Done
The following packages will be upgraded:
  apt apt-transport-https apt-utils base-files libapt-inst2.0 libapt-pkg5.0 libpam-systemd libperl5.24 libpolkit-agent-1-0
  libpolkit-backend-1-0 libpolkit-gobject-1-0 libssl1.0.2 libssl1.1 libsystemd0 libudev1 libwbclient0 openssl perl perl-base
  perl-modules-5.24 policykit-1 python-rpi.gpio raspberrypi-sys-mods raspi-config samba-common systemd systemd-sysv udev
28 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Need to get 17.0 MB of archives.
After this operation, 134 kB of additional disk space will be used.
Get:1 http://archive.raspberrypi.org/debian stretch/main armhf python-rpi.gpio armhf 0.6.5~stretch-1 [23.5 kB]
Get:2 http://archive.raspberrypi.org/debian stretch/main armhf raspberrypi-sys-mods armhf 20181127 [10.2 kB]
Get:3 http://archive.raspberrypi.org/debian stretch/ui armhf raspi-config all 20181118 [20.2 kB]
Get:4 http://mirror.de.leaseweb.net/raspbian/raspbian stretch/main armhf base-files armhf 9.9+rpi1+deb9u6 [67.5 kB]
Get:5 http://mirror.de.leaseweb.net/raspbian/raspbian stretch/main armhf libperl5.24 armhf 5.24.1-3+deb9u5 [2837 kB]
Get:6 http://mirror.de.leaseweb.net/raspbian/raspbian stretch/main armhf perl armhf 5.24.1-3+deb9u5 [219 kB]
Get:7 http://mirror.de.leaseweb.net/raspbian/raspbian stretch/main armhf perl-base armhf 5.24.1-3+deb9u5 [1198 kB]
Get:8 http://mirror.de.leaseweb.net/raspbian/raspbian stretch/main armhf perl-modules-5.24 all 5.24.1-3+deb9u5 [2722 kB]
Get:9 http://mirror.de.leaseweb.net/raspbian/raspbian stretch/main armhf libapt-pkg5.0 armhf 1.4.9 [833 kB]
Get:10 http://mirror.de.leaseweb.net/raspbian/raspbian stretch/main armhf libapt-inst2.0 armhf 1.4.9 [190 kB]
Get:11 http://mirror.de.leaseweb.net/raspbian/raspbian stretch/main armhf apt armhf 1.4.9 [1190 kB]
Get:12 http://mirror.de.leaseweb.net/raspbian/raspbian stretch/main armhf apt-utils armhf 1.4.9 [392 kB]
Get:13 http://mirror.de.leaseweb.net/raspbian/raspbian stretch/main armhf systemd-sysv armhf 232-25+deb9u8 [81.8 kB]
Get:14 http://mirror.de.leaseweb.net/raspbian/raspbian stretch/main armhf libpam-systemd armhf 232-25+deb9u8 [174 kB]
Get:15 http://mirror.de.leaseweb.net/raspbian/raspbian stretch/main armhf libsystemd0 armhf 232-25+deb9u8 [259 kB]
Get:16 http://mirror.de.leaseweb.net/raspbian/raspbian stretch/main armhf systemd armhf 232-25+deb9u8 [2221 kB]
Get:17 http://mirror.de.leaseweb.net/raspbian/raspbian stretch/main armhf udev armhf 232-25+deb9u8 [1072 kB]
Get:18 http://mirror.de.leaseweb.net/raspbian/raspbian stretch/main armhf libudev1 armhf 232-25+deb9u8 [120 kB]
Get:19 http://mirror.de.leaseweb.net/raspbian/raspbian stretch/main armhf libwbclient0 armhf 2:4.5.12+dfsg-2+deb9u4 [121 kB]
Get:20 http://mirror.de.leaseweb.net/raspbian/raspbian stretch/main armhf samba-common all 2:4.5.12+dfsg-2+deb9u4 [172 kB]
Get:21 http://mirror.de.leaseweb.net/raspbian/raspbian stretch/main armhf libssl1.0.2 armhf 1.0.2q-1~deb9u1 [898 kB]
Get:22 http://mirror.de.leaseweb.net/raspbian/raspbian stretch/main armhf libssl1.1 armhf 1.1.0j-1~deb9u1 [1115 kB]
Get:23 http://mirror.de.leaseweb.net/raspbian/raspbian stretch/main armhf apt-transport-https armhf 1.4.9 [166 kB]
Get:24 http://mirror.de.leaseweb.net/raspbian/raspbian stretch/main armhf policykit-1 armhf 0.105-18+deb9u1 [61.7 kB]
Get:25 http://mirror.de.leaseweb.net/raspbian/raspbian stretch/main armhf libpolkit-agent-1-0 armhf 0.105-18+deb9u1 [22.3 kB]
Get:26 http://mirror.de.leaseweb.net/raspbian/raspbian stretch/main armhf libpolkit-backend-1-0 armhf 0.105-18+deb9u1 [39.8 kB]
Get:27 http://mirror.de.leaseweb.net/raspbian/raspbian stretch/main armhf libpolkit-gobject-1-0 armhf 0.105-18+deb9u1 [37.8 kB]
Get:28 http://mirror.de.leaseweb.net/raspbian/raspbian stretch/main armhf openssl armhf 1.1.0j-1~deb9u1 [712 kB]
Fetched 17.0 MB in 4s (3699 kB/s) 
apt-listchanges: Reading changelogs...
Preconfiguring packages ...
libssl1.0.2 failed to preconfigure, with exit status 1
libssl1.1 failed to preconfigure, with exit status 1
(Reading database ... 34605 files and directories currently installed.)
Preparing to unpack .../base-files_9.9+rpi1+deb9u6_armhf.deb ...
Unpacking base-files (9.9+rpi1+deb9u6) over (9.9+rpi1+deb9u4) ...
Setting up base-files (9.9+rpi1+deb9u6) ...
Installing new version of config file /etc/debian_version ...
(Reading database ... 34605 files and directories currently installed.)
Preparing to unpack .../libperl5.24_5.24.1-3+deb9u5_armhf.deb ...
Unpacking libperl5.24:armhf (5.24.1-3+deb9u5) over (5.24.1-3+deb9u4) ...
Preparing to unpack .../perl_5.24.1-3+deb9u5_armhf.deb ...
Unpacking perl (5.24.1-3+deb9u5) over (5.24.1-3+deb9u4) ...
Preparing to unpack .../perl-base_5.24.1-3+deb9u5_armhf.deb ...
Unpacking perl-base (5.24.1-3+deb9u5) over (5.24.1-3+deb9u4) ...
Setting up perl-base (5.24.1-3+deb9u5) ...
(Reading database ... 34605 files and directories currently installed.)
Preparing to unpack .../perl-modules-5.24_5.24.1-3+deb9u5_all.deb ...
Unpacking perl-modules-5.24 (5.24.1-3+deb9u5) over (5.24.1-3+deb9u4) ...
Preparing to unpack .../libapt-pkg5.0_1.4.9_armhf.deb ...
Unpacking libapt-pkg5.0:armhf (1.4.9) over (1.4.8) ...
Setting up libapt-pkg5.0:armhf (1.4.9) ...
(Reading database ... 34605 files and directories currently installed.)
Preparing to unpack .../libapt-inst2.0_1.4.9_armhf.deb ...
Unpacking libapt-inst2.0:armhf (1.4.9) over (1.4.8) ...
[... 32 lines cut ...]
Preparing to unpack .../05-policykit-1_0.105-18+deb9u1_armhf.deb ...
Unpacking policykit-1 (0.105-18+deb9u1) over (0.105-18) ...
Preparing to unpack .../06-libpolkit-agent-1-0_0.105-18+deb9u1_armhf.deb ...
Unpacking libpolkit-agent-1-0:armhf (0.105-18+deb9u1) over (0.105-18) ...
Preparing to unpack .../07-libpolkit-backend-1-0_0.105-18+deb9u1_armhf.deb ...
Unpacking libpolkit-backend-1-0:armhf (0.105-18+deb9u1) over (0.105-18) ...
Preparing to unpack .../08-libpolkit-gobject-1-0_0.105-18+deb9u1_armhf.deb ...
Unpacking libpolkit-gobject-1-0:armhf (0.105-18+deb9u1) over (0.105-18) ...
Preparing to unpack .../09-openssl_1.1.0j-1~deb9u1_armhf.deb ...
Unpacking openssl (1.1.0j-1~deb9u1) over (1.1.0f-3+deb9u2) ...
Preparing to unpack .../10-python-rpi.gpio_0.6.5~stretch-1_armhf.deb ...
Unpacking python-rpi.gpio (0.6.5~stretch-1) over (0.6.3~stretch-1) ...
Preparing to unpack .../11-raspberrypi-sys-mods_20181127_armhf.deb ...
Error while loading /var/lib/dpkg/tmp.ci/preinst: Exec format error
dpkg: error processing archive /tmp/apt-dpkg-install-CHEtuw/11-raspberrypi-sys-mods_20181127_armhf.deb (--unpack):
 subprocess new pre-installation script returned error exit status 1
Error while loading /var/lib/dpkg/info/raspberrypi-sys-mods.postinst: Exec format error
dpkg: error while cleaning up:
 subprocess installed post-installation script returned error exit status 1
Preparing to unpack .../12-raspi-config_20181118_all.deb ...
Unpacking raspi-config (20181118) over (20180518) ...
Errors were encountered while processing:
 /tmp/apt-dpkg-install-CHEtuw/11-raspberrypi-sys-mods_20181127_armhf.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

geistesk, 2019-01-23:
Could you please paste your `Pifile`?

gh0st42, 2019-01-23:
```

FROM /input/2018-11-13-raspbian-stretch-lite.img
TO /output/buschfunk.img
 
PUMP 200
 
ENABLE_UART
RUN uname -a
RUN apt-get update
echo "UPGRADING"
# The following would be the correct way... 
#RUN bash -c "DEBIAN_FRONTEND=noninteractive apt-get -y dist-upgrade"

RUN apt-get -y dist-upgrade
```

geistesk, 2019-01-23:
Is the `DEBIAN_FRONTEND=noninteractive` environment variable for a *dist-upgrade* essential?

geistesk, 2019-01-29:
After taking a first look into this issue, the `raspberrypi-sys-mods` package seems to be the show stopper. The following setup demonstrates this package's failure in our emulation:

```
cat Upgrade.Pifile
FROM 2018-11-13-raspbian-stretch-lite.img

PUMP 100

ENABLE_UART

# Failing package is raspberrypi-sys-mods_20181127_armhf.deb
RUN apt-get update
RUN apt-mark hold raspberrypi-sys-mods
RUN bash -c 'DEBIAN_FRONTEND=noninteractive apt-get -y dist-upgrade'
```

After booting the created image and *unholding* this package, the *dist-upgrade* works flawlessly.

commit 6d881095 (2019-01-29): Fixed #3 by replacing proot with chroot
closed 2019-01-29 by geistesk