Nature40/pimod #6: Infer image name from Pifile name

jonashoechst, 2019-01-24:
When not configured explicitly (via TO ...) the name of the resulting image can also be referred from the Pifile, e.g. BaumBox.Pifile would result in the image being called BaumBox.img.

gh0st42, 2019-01-24:
If removing TO is desired then removing FROM might also be a good idea. If no FROM is given just expect a default image (e.g. `/raspbian.img`). This can be easily overmapped by a volume. Therefore, just provide the input image as parameter to the wrapper script and omit changing the FROM line every time you switch between different input images.

jonashoechst, 2019-01-24:
To clarify: I was not talking about removing TO, but extending the current functionality. Currently, if TO is not specified, the resulting image is written to rpi.img.

gh0st42, 2019-01-24:
Similarly FROM could be optional, no need to remove it completely since it is already implemented.
Would make preparing several images with same setup easier (e.g. raspbian-lite and then an raspbian-full)

commit 15d8af2f (2019-01-29): Infer output/TO-name from named Pifile

This commit closes #6 by inspecting the filename of the given Pifile. If
the Pifile is named "foobar.Pifile" and no further `TO` was specified,
an image named "foobar.img" will be generated.

> $ cat Foobar.Pifile
> FROM 2018-11-13-raspbian-stretch-lite.img
> RUN uname -m
>
> $ ./pimod-docker.sh \
>   Foobar.Pifile \
>   ~/Downloads/2018-11-13-raspbian-stretch-lite.img \
>   Foobar.img
> ### FROM 2018-11-13-raspbian-stretch-lite.img
> ### TO Foobar.img
> ### RUN uname -m
> armv7l
> loop deleted : /dev/loop1
closed 2019-01-29 by geistesk