Nature40/pimod #7: Provide an option for Pifile inheritance

jonashoechst, 2019-01-24:
As an addition to the FROM definition I suggest to introduce an PARENT statement.

When a Pifile stars with an PARENT statement, pimod first creates the image defined in the parent Pifile and afterwards continues executing the Pifile.

gh0st42, 2019-01-24:
Mmmh... something smells like docker ;)
Isn't it enough to reference your previously generated image in FROM? Otherwise one could just source another Pifile in the beginning. Should work, shouldn't it?
If intermediate images are a thing then setting a temporary working directory for image generation might also be a good idea..

jonashoechst, 2019-01-29:
Actually either using a Base Image with `FROM` or `source` of another Pifile are two options that work just fine for me. In my opinion this possibilities should be documented.

closed 2019-01-29 by jonashoechst