Nature40/pimod #10: PUMP not working?

i5o, 2019-02-14:
Hello. I'm trying to get to use pimod, but my first test already failed :)
Here is my Pifile
```
FROM image/2018-11-13-raspbian-stretch.img

PUMP 1500

RUN bash -c 'echo "${CHROOT_MOUNT}"'
```

and here is the output I get, is this normal? I tried running as non-sudo user:
```
ignacio@cat:~/pimod$ ./pimod.sh treehouses.Pifile 
### FROM image/2018-11-13-raspbian-stretch.img
### TO treehouses.img
### PUMP 1500
1500+0 records in
1500+0 records out
1572864000 bytes (1,6 GB, 1,5 GiB) copied, 1,00085 s, 1,6 GB/s
/dev/mapper/control: open failed: Permission denied
Failure to communicate with kernel device-mapper driver.
Incompatible libdevmapper 1.02.145 (2017-11-03) and kernel driver (unknown version).
device mapper prerequisites not met
e2fsck 1.44.4 (18-Aug-2018)
e2fsck: No such file or directory while trying to open /dev/mapper/p2
Possibly non-existent device?
ignacio@cat:~/pimod$ 
```

then with sudo
```
ignacio@cat:~/pimod$ sudo ./pimod.sh treehouses.Pifile 
### FROM image/2018-11-13-raspbian-stretch.img
### TO treehouses.img
### PUMP 1500
1500+0 records in
1500+0 records out
1572864000 bytes (1,6 GB, 1,5 GiB) copied, 0,803511 s, 2,0 GB/s
loop16p1 is already in use, and a partition of a different device
loop16p2 is already in use, and a partition of a different device
e2fsck 1.44.4 (18-Aug-2018)
e2fsck: No such file or directory while trying to open /dev/mapper/p2
Possibly non-existent device?
ignacio@cat:~/pimod$
```

also docker
```
ignacio@cat:~/pimod$ ./pimod-docker.sh treehouses.Pifile image/2018-11-13-raspbian-stretch.img image/aa.img
### FROM 2018-11-13-raspbian-stretch.img
### TO treehouses.img
### PUMP 1500
1500+0 records in
1500+0 records out
1572864000 bytes (1.6 GB, 1.5 GiB) copied, 4.42665 s, 355 MB/s
e2fsck 1.43.4 (31-Jan-2017)
e2fsck: No such file or directory while trying to open /dev/mapper/add map loop16p1 (253:0): 0 89854 linear 7:16 8192p2
Possibly non-existent device?
ignacio@cat:~/pimod
```

Just wondering if this is normal or to check if I'm doing something wrong

thank you

geistesk, 2019-02-14:
Thanks for trying pimod and please excuse the troubles.

Based on your first and the following logs, the problem lies with the execution of pimod's chroot part as an unprivileged user. However, it seems like a loop device was created, which is now causing problems.

An easy fix would be:
```shell
# check if there are any loop devices
losetup -a

# delete all orphaned loop devices (replace loop0 with your devices)
sudo losetup -d /dev/loop0

# just clean up
sudo dmsetup remove_all

# Check again and, if anything is present, try deleting them again.
# If everything fails, reboot your machine ¯\_(ツ)_/¯
```

Afterwards, running your `Pifile` as a privileged user (like `root` or inside Docker) should work.

Please consider, the `CHROOT_MOUNT` variable is used inside the `pimod.sh` script, but should not be passed inside the chroot. Therefore your `echo` - executed inside the chroot - will be empty.

i5o, 2019-02-20:
So I just started a clean session (rebooted my system and started from scratch) and here is my whole output, not sure if this error is relevant

`e2fsck: No such file or directory while trying to open /dev/mapper/add map loop17p1 (253:0): 0 89854 linear 7:17 8192p2`

```
ignacio@cat:~/pimod$ losetup -a
/dev/loop1: []: (/var/lib/snapd/snaps/core_6350.snap)
/dev/loop8: []: (/var/lib/snapd/snaps/gnome-logs_45.snap)
/dev/loop15: []: (/var/lib/snapd/snaps/gnome-system-monitor_57.snap)
/dev/loop6: []: (/var/lib/snapd/snaps/gnome-characters_139.snap)
/dev/loop13: []: (/var/lib/snapd/snaps/gnome-characters_124.snap)
/dev/loop4: []: (/var/lib/snapd/snaps/gnome-calculator_260.snap)
/dev/loop11: []: (/var/lib/snapd/snaps/core_6405.snap)
/dev/loop2: []: (/var/lib/snapd/snaps/core_6259.snap)
/dev/loop0: []: (/var/lib/snapd/snaps/gtk-common-themes_818.snap)
/dev/loop9: []: (/var/lib/snapd/snaps/gnome-3-26-1604_78.snap)
/dev/loop16: []: (/var/lib/snapd/snaps/gtk-common-themes_701.snap)
/dev/loop7: []: (/var/lib/snapd/snaps/postman_81.snap)
/dev/loop14: []: (/var/lib/snapd/snaps/gnome-3-26-1604_70.snap)
/dev/loop5: []: (/var/lib/snapd/snaps/gnome-3-26-1604_74.snap)
/dev/loop12: []: (/var/lib/snapd/snaps/postman_80.snap)
/dev/loop3: []: (/var/lib/snapd/snaps/gtk-common-themes_1122.snap)
/dev/loop10: []: (/var/lib/snapd/snaps/gnome-calculator_238.snap)
ignacio@cat:~/pimod$ 
ignacio@cat:~/pimod$ sudo dmsetup remove_all
[sudo] password for ignacio: 
ignacio@cat:~/pimod$ ./pimod.sh treehouses.Pifile 
### FROM image/2018-11-13-raspbian-stretch.img
### TO treehouses.img
### PUMP 1500
1500+0 records in
1500+0 records out
1572864000 bytes (1,6 GB, 1,5 GiB) copied, 0,973376 s, 1,6 GB/s
/dev/mapper/control: open failed: Permission denied
Failure to communicate with kernel device-mapper driver.
Incompatible libdevmapper 1.02.145 (2017-11-03) and kernel driver (unknown version).
device mapper prerequisites not met
e2fsck 1.44.4 (18-Aug-2018)
e2fsck: No such file or directory while trying to open /dev/mapper/p2
Possibly non-existent device?
ignacio@cat:~/pimod$ losetup -a
/dev/loop1: []: (/var/lib/snapd/snaps/core_6350.snap)
/dev/loop8: []: (/var/lib/snapd/snaps/gnome-logs_45.snap)
/dev/loop15: []: (/var/lib/snapd/snaps/gnome-system-monitor_57.snap)
/dev/loop6: []: (/var/lib/snapd/snaps/gnome-characters_139.snap)
/dev/loop13: []: (/var/lib/snapd/snaps/gnome-characters_124.snap)
/dev/loop4: []: (/var/lib/snapd/snaps/gnome-calculator_260.snap)
/dev/loop11: []: (/var/lib/snapd/snaps/core_6405.snap)
/dev/loop2: []: (/var/lib/snapd/snaps/core_6259.snap)
/dev/loop0: []: (/var/lib/snapd/snaps/gtk-common-themes_818.snap)
/dev/loop9: []: (/var/lib/snapd/snaps/gnome-3-26-1604_78.snap)
/dev/loop16: []: (/var/lib/snapd/snaps/gtk-common-themes_701.snap)
/dev/loop7: []: (/var/lib/snapd/snaps/postman_81.snap)
/dev/loop14: []: (/var/lib/snapd/snaps/gnome-3-26-1604_70.snap)
/dev/loop5: []: (/var/lib/snapd/snaps/gnome-3-26-1604_74.snap)
/dev/loop12: []: (/var/lib/snapd/snaps/postman_80.snap)
/dev/loop3: []: (/var/lib/snapd/snaps/gtk-common-themes_1122.snap)
/dev/loop10: []: (/var/lib/snapd/snaps/gnome-calculator_238.snap)
ignacio@cat:~/pimod$ sudo dmsetup remove_all
ignacio@cat:~/pimod$ sudo ./pimod.sh treehouses.Pifile 
### FROM image/2018-11-13-raspbian-stretch.img
### TO treehouses.img
### PUMP 1500
1500+0 records in
1500+0 records out
1572864000 bytes (1,6 GB, 1,5 GiB) copied, 0,912215 s, 1,7 GB/s
e2fsck 1.44.4 (18-Aug-2018)
e2fsck: No such file or directory while trying to open /dev/mapper/add map loop17p1 (253:0): 0 89854 linear 7:17 8192p2
Possibly non-existent device?
ignacio@cat:~/pimod$ 
```

am I doing something wrong about it?

jonashoechst, 2019-09-04:
Could you please try to execute pima.sh as root using sudo? 

As you see, problems start to occur, when you `/dev/mapper/control` can't be opened:

```
ignacio@cat:~/pimod$ ./pimod.sh treehouses.Pifile 
### FROM image/2018-11-13-raspbian-stretch.img
### TO treehouses.img
### PUMP 1500
1500+0 records in
1500+0 records out
1572864000 bytes (1,6 GB, 1,5 GiB) copied, 0,973376 s, 1,6 GB/s
/dev/mapper/control: open failed: Permission denied
...
```

jonashoechst, 2019-11-01:
Closed due to no further activity. Feel free to reopen!

closed 2019-11-01 by jonashoechst