Nature40/pimod #11: Code editor support for Pifile

empeje, 2019-02-18:
clear enough.

aniongithub, 2022-01-24:
I could give this a shot and write an extension for VS Code.

empeje, 2022-01-31:
Let me know if I can also contribute to your VS Code extension @aniongithub

disaac, 2023-03-29:
@aniongithub @empeje  I threw together a vscode syntax/language extension that Im using I just pushed it up. I did this very fast and just basically merged together Bash/Shell and Dockerfile tmlanguage grammar. Removing the unsupported Docker keywords from the pattern array. Works if you put it in your ~/.vscode/extensions folder. Needs work since I just did threw it together to get basic syntax support. 

Thanks so much for the awesome project btw Nature40

[vscode-pifile](https://github.com/disaac/vscode-pifile)

oxzi, 2023-04-07:
@disaac: This is really cool, thanks a lot!