Nature40/pimod #14: Implement INSTALL by using rsync

jonashoechst, 2019-11-06:
Hidden files are still not copied, when installing whole folder. I suggest to use rsync for installation to install those files as well.

geistesk, 2019-11-20:
This potential problem has already been widely discussed online, e.g. [here](https://superuser.com/q/61611). One possibility would be to copy directories with `--no-target-directory`.

In addition, I find that setting permissions for whole directories would not make so much sense. Maybe we should remove this option from `INSTALL`, especially since it can be executed via a simple `RUN chmod …`. However, this would also be another issue or none at all.

oxzi, 2021-10-04:
@jonashoechst: Is this issue still present? I have tried it out here.

```
$ cat openwrt.Pifile
FROM http://downloads.openwrt.org/releases/18.06.5/targets/brcm2708/bcm2710/openwrt-18.06.5-brcm2708-bcm2710-rpi-3-ext4-factory.img.gz
TO hiddentest.img

HOST mkdir -p /tmp/hiddentest/.foo/.bar
HOST touch /tmp/hiddentest/.foo/.bar/.buz

INSTALL /tmp/hiddentest /root/hiddentest

RUN find /root/hiddentest

$ docker-compose run pimod pimod.sh openwrt.Pifile
. . .
### HOST mkdir -p /tmp/hiddentest/.foo/.bar
### HOST touch /tmp/hiddentest/.foo/.bar/.buz
### INSTALL /tmp/hiddentest /root/hiddentest
### RUN find /root/hiddentest
/root/hiddentest
/root/hiddentest/.foo
/root/hiddentest/.foo/.bar
/root/hiddentest/.foo/.bar/.buz
. . .
```

jonashoechst, 2021-10-04:
Thanks @oxzi for getting back. The MWE presented does indeed install hidden files and folders, hence the issue seems to be fixed.

closed 2021-10-04 by jonashoechst