Nature40/pimod #16: Derive PATH from the guest OS

jonashoechst, 2019-11-14:
Currently, the PATH variable is just copied from the source OS. This works when modifying Debian-based OSss on Debian but fails, when targeting some other OSes. 

We should generalize this and maybe provide an option which file to source.

geistesk, 2019-11-14:
Guessing the right `PATH` could be difficult or even unfeasible, e.g., using a plain NixOS image. In general, different target and source systems are hard.

Even sourcing some `.bashrc`, `.zshrc`, or `.whateverrc` file could result in lots of unintended side effects. For example, overwriting other variables or executing unintended code.

Thus, I would recommend adding a `PATH` command which just overlays the `PATH` variable for the `RUN` command.

oxzi, 2021-10-04:
As previously discussed, this only works if the guest system uses the same `PATH` as the host.
However, if necessary, one can `source` the `.bashrc` within the image.

Otherwise, there is the `PATH` command within pimod to overwrite the guest's `PATH`.

closed 2021-10-04 by oxzi