Nature40/pimod #18: INPLACE read only mode

jonashoechst, 2019-11-14:
I suggest to have INPLACE mount the image read only. Thereby one can test stuff while making sure, that no contents of the original image is modified. 

We can discuss on this.

geistesk, 2019-11-14:
In my opinion, the `INPLACE` command should make the changes in place. Additionally a read-only flag could be added or another command, e.g., `DRYRUN`.

jonashoechst, 2019-11-14:
Actually the uses cases are pretty rare. For testing, as we do currently, a read-only INPLACE would be nice, since I don't want my image to be destroyed. However people actually using pimod use it to modify images, having a read only option is kind of contradictory.

Nevermind.

closed 2019-11-14 by jonashoechst