Nature40/pimod #22: OpenWRT test fails after implementing PATH feature

jonashoechst, 2019-11-14:
see https://github.com/Nature40/pimod-ci/commit/d220adcaff7dcb129f6315da991ee3e1c6949b69/checks?check_suite_id=311137370

geistesk, 2019-11-15:
I just tested the following Pifile successfully, which looks identically to the failed job.

```
FROM https://downloads.openwrt.org/releases/18.06.4/targets/brcm2708/bcm2710/openwrt-18.06.4-brcm2708-bcm2710-rpi-3-ext4-factory.img.gz

RUN cat /etc/os-release
```

Furthermore, the error looks like a failed `mount -a` to me. The guest fails to execute both the internal `mount` and the `cat` command. I'm not quite sure what is broken there. However, I cannot reproduce this behavior.

jonashoechst, 2019-11-15:
I started to investigate using the interactive debug option, however I do not quite understand, why the chroot environment does not work in this case.

jonashoechst, 2019-11-15:
Something is deliberately broken here, not even Raspbian does work ATM. It fails for me, fails in CI.
Please have a look at this @geistesk.

jonashoechst, 2019-11-16:
The bug was introduced by moving the QEMU binaries to a temporary folder which is added to the PATH. However, the path of the binaries is resolved when enabling the architecture using update-binfmts.

The enabled path is available on the host system but not on the guest.

commit 5cf99fa0 (2019-11-16): use identical path for qemu binaries on host and quest (fixes #22)
closed 2019-11-16 by jonashoechst