Nature40/pimod #24: pimod fails on btrfs mounts

jonashoechst, 2019-11-15:
btrfs maintains su bvolumes inside of containers. When mounting a container, the sub volumes are accessible as a folder. 

pimod is currently not able to work on operating systems booting from btrfs, as we are only mounting the container not the submodule. 

Suggestion for a simple fix:
 - If a btrfs container is found and one sub volume is inside, mount this sub volume.

RafalSkolasinski, 2021-01-07:
Just tried the example and it fails, wonder if it is because of this issue?
I am running Tumbleweed with BTRFS...

```
### FROM http://downloads.openwrt.org/releases/18.06.5/targets/brcm2708/bcm2710/openwrt-18.06.5-brcm2708-bcm2710-rpi-3-ext4-factory.img.gz
Using cache: .cache/downloads.openwrt.org/releases/18.06.5/targets/brcm2708/bcm2710/openwrt-18.06.5-brcm2708-bcm2710-rpi-3-ext4-factory.img.gz
Moving temporary /tmp/tmp.KOL1PvarE2 to examples/RPi-OpenWRT.img
### PUMP 10M
1+0 records in
1+0 records out
10485760 bytes (10 MB, 10 MiB) copied, 0.00342355 s, 3.1 GB/s

Welcome to fdisk (util-linux 2.36.1).
Changes will remain in memory only, until you decide to write them.
Be careful before using the write command.

Command (m for help): Partition number (1,2, default 2): 
Partition 2 has been deleted.

Command (m for help): Partition type
   p   primary (1 primary, 0 extended, 3 free)
   e   extended (container for logical partitions)
Select (default p): Partition number (2-4, default 2): First sector (2048-602111, default 2048): Last sector, +/-sectors or +/-size{K,M,G,T,P} (57344-602111, default 602111): Partition #2 contains a ext4 signature.

Created a new partition 2 of type 'Linux' and of size 266 MiB.

Command (m for help): 
The partition table has been altered.
Syncing disks.

add map loop4p1 (254:9): 0 40960 linear 7:4 8192
add map loop4p2 (254:10): 0 544768 linear 7:4 57344
e2fsck 1.45.6 (20-Mar-2020)
Pass 1: Checking inodes, blocks, and sizes
Pass 2: Checking directory structure
Pass 3: Checking directory connectivity
Pass 4: Checking reference counts
Pass 5: Checking group summary information
Padding at end of inode bitmap is not set. Fix? yes

/dev/mapper/loop4p2: ***** FILE SYSTEM WAS MODIFIED *****
/dev/mapper/loop4p2: 1264/16384 files (0.0% non-contiguous), 4278/65536 blocks
```

RafalSkolasinski, 2021-01-07:
Actually, got the same on another machine that runs on ext4. Gonna open another issue.

oxzi, 2021-10-04:
This might be kind of tricky, as other filesystems then ext3 or ex4 also influence the `PUMP` command.
Of course, we could first check the filesystem on the partition and alter both the `mount` as well as the filesystem resizing.

@jonashoechst: Do you remember which image was using btrfs?

jonashoechst, 2021-10-04:
Unfortunately I don't remember which image or distribution that was. I think that btrfs is not very common on single board computers and would therefore close the issue.

If the problem should come up again, the issue can be reopened.

closed 2021-10-04 by jonashoechst