Nature40/pimod #29: script stops unexpectedly: FILE SYSTEM WAS MODIFIED

RafalSkolasinski, 2021-01-07:
When trying to run the example it stops with following logs
```bash
### FROM http://downloads.openwrt.org/releases/18.06.5/targets/brcm2708/bcm2710/openwrt-18.06.5-brcm2708-bcm2710-rpi-3-ext4-factory.img.gz
Fetching remote: http://downloads.openwrt.org/releases/18.06.5/targets/brcm2708/bcm2710/openwrt-18.06.5-brcm2708-bcm2710-rpi-3-ext4-factory.img.gz
--2021-01-07 00:07:52--  http://downloads.openwrt.org/releases/18.06.5/targets/brcm2708/bcm2710/openwrt-18.06.5-brcm2708-bcm2710-rpi-3-ext4-factory.img.gz
Resolving downloads.openwrt.org (downloads.openwrt.org)... 168.119.138.211, 2a01:4f8:251:321::2
Connecting to downloads.openwrt.org (downloads.openwrt.org)|168.119.138.211|:80... connected.
HTTP request sent, awaiting response... 301 Moved Permanently
Location: https://downloads.openwrt.org/releases/18.06.5/targets/brcm2708/bcm2710/openwrt-18.06.5-brcm2708-bcm2710-rpi-3-ext4-factory.img.gz [following]
--2021-01-07 00:07:53--  https://downloads.openwrt.org/releases/18.06.5/targets/brcm2708/bcm2710/openwrt-18.06.5-brcm2708-bcm2710-rpi-3-ext4-factory.img.gz
Connecting to downloads.openwrt.org (downloads.openwrt.org)|168.119.138.211|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 9268927 (8.8M) [application/octet-stream]
Saving to: '.cache/downloads.openwrt.org/releases/18.06.5/targets/brcm2708/bcm2710/openwrt-18.06.5-brcm2708-bcm2710-rpi-3-ext4-factory.img.gz'

     0K ........                           100%  842K=11s

2021-01-07 00:08:04 (842 KB/s) - '.cache/downloads.openwrt.org/releases/18.06.5/targets/brcm2708/bcm2710/openwrt-18.06.5-brcm2708-bcm2710-rpi-3-ext4-factory.img.gz' saved [9268927/9268927]

Moving temporary /tmp/tmp.yerr1r1Ppy to examples/RPi-OpenWRT.img
### PUMP 10M
1+0 records in
1+0 records out
10485760 bytes (10 MB, 10 MiB) copied, 0.00650014 s, 1.6 GB/s

Welcome to fdisk (util-linux 2.36.1).
Changes will remain in memory only, until you decide to write them.
Be careful before using the write command.

Command (m for help): Partition number (1,2, default 2): 
Partition 2 has been deleted.

Command (m for help): Partition type
   p   primary (1 primary, 0 extended, 3 free)
   e   extended (container for logical partitions)
Select (default p): Partition number (2-4, default 2): First sector (2048-602111, default 2048): Last sector, +/-sectors or +/-size{K,M,G,T,P} (57344-602111, default 602111): Partition #2 contains a ext4 signature.

Created a new partition 2 of type 'Linux' and of size 266 MiB.

Command (m for help): 
The partition table has been altered.
Syncing disks.

add map loop0p1 (253:3): 0 40960 linear 7:0 8192
add map loop0p2 (253:4): 0 544768 linear 7:0 57344
e2fsck 1.45.6 (20-Mar-2020)
Pass 1: Checking inodes, blocks, and sizes
Pass 2: Checking directory structure
Pass 3: Checking directory connectivity
Pass 4: Checking reference counts
Pass 5: Checking group summary information
Padding at end of inode bitmap is not set. Fix? yes

/dev/mapper/loop0p2: ***** FILE SYSTEM WAS MODIFIED *****
/dev/mapper/loop0p2: 1264/16384 files (0.0% non-contiguous), 4278/65536 blocks
```

RafalSkolasinski, 2021-01-07:
Any idea what can be going wrong? @oxzi @jonashoechst

closed 2021-01-17 by jonashoechst

jonashoechst, 2021-01-17:
Thank you for opening the issue. Turned out, e2fsck's return codes were not evaluated correctly, hence pimod failed on a successful file system repair. It is fixed in the master branch and is part of v0.2.1.

commit 3e1028ed (2021-01-17): bugfix: successful e2fsck lead to failing build

- e2fsck's return codes 1 and 2 refer to found but corrected errors,
  hence a successful repair lead to abortion
- the bug was addressed by manually checking the return code
- closes #29