Nature40/pimod #31: When FROM download fails, subsequent commands fail because of using cache

aniongithub, 2021-04-26:
First run for an unsuccessful download, see issue [30](https://github.com/Nature40/pimod/issues/30)

```
Fetching remote: https://downloads.raspberrypi.org/raspios_lite_armhf/images/raspios_lite_armhf-2021-03-25/2021-03-04-raspios-buster-armhf-lite.zip
--2021-04-26 17:15:03--  https://downloads.raspberrypi.org/raspios_lite_armhf/images/raspios_lite_armhf-2021-03-25/2021-03-04-raspios-buster-armhf-lite.zip%0D
Resolving downloads.raspberrypi.org (downloads.raspberrypi.org)... 46.235.227.39, 176.126.240.84, 93.93.135.117, ...
Connecting to downloads.raspberrypi.org (downloads.raspberrypi.org)|46.235.227.39|:443... connected.
HTTP request sent, awaiting response... 404 Not Found
2021-04-26 17:15:04 ERROR 404: Not Found.

```

second try uses cache even though the first download did not succeed?

```
docker run --rm --network=host --privileged -v $PWD:/pimod pimod pimod.sh examples/Rpi-Docker.Pifile
### FROM https://downloads.raspberrypi.org/raspios_lite_armhf/images/raspios_lite_armhf-2021-03-25/2021-03-04-raspios-buster-armhf-lite.zip
Using cache: .cache/downloads.raspberrypi.org/raspios_lite_armhf/images/raspios_lite_armhf-2021-03-25/2021-03-04-raspios-buster-armhf-lite.zip
### Error: Unknown MIME inode/x-empty
```

I'm also seeing that the cache has an entry for 

```
ls -al .cache/downloads.raspberrypi.org/raspios_lite_armhf/images/raspios_lite_armhf-2021-03-25/
total 0
drwxrwxrwx 1 ani ani 4096 Apr 26 10:20  .
drwxrwxrwx 1 ani ani 4096 Apr 26 10:20  ..
-rwxrwxrwx 1 ani ani    0 Apr 26 10:20 '2021-03-04-raspios-buster-armhf-lite.zip'$'\r'
```

Not sure how to use this with any official pi image without downloading it first...

oxzi, 2021-04-26:
Thanks for reporting.

Caching an invalid image file is definitely a bug where we should look into. Thanks.

However, the filename looks suspicious. Could you please provide your Pifile in its original encoding? The `\r` might be due to some OS-specific newline representation. Please keep in mind that a Pifile is nothing else than a Bash script.

> Not sure how to use this with any official pi image without downloading it first...

It's also possible to provide an image located on your local storage. `FROM` works with both local and remote files, [documentation](https://github.com/Nature40/pimod#from-path_to_image-from-url).

aniongithub, 2021-04-26:
I would like to use a downloaded image to enable certain workflows, using a locally downloaded file is not the best option for me. I could implement a workaround by `wget`-ing the file by myself, before invoking it  with a local image reference, but since this exists I wanted to leverage it for a fully online workflow.

aniongithub, 2021-04-28:
This issue, however is still valid. When a download fails, `pimod.sh` still tries to use the partially downloaded image and errors out.

oxzi, 2021-05-02:
Thanks again for reporting this issue.

I added a check to the download which cleans up after a failed `wget`. However, this only solves future issues. To cleanup invalid entries within your cache, please purge it.

commit cc8423e3 (2021-05-02): from_remote_fetch: delete image if download failed

Do not save an incorrectly downloaded image in the cache.

This closes #31.
closed 2021-05-02 by oxzi