Nature40/pimod #34: How can we modify pimod to work with Jetson SD card images?

aniongithub, 2021-09-23:
Pimod is awesome and could be even more useful if we could use it work with Jetson Nano/Xavier images.

See an example of a custom-built Jetson image here: https://github.com/aniongithub/jetson-nano-image/releases

While this currently uses Ansible, I thought it would be cool if we could use Pimod to build on top of the custom minimal image generated. Sort of like how Pimod is meant to customize a raspbian-lite image.

Any pointers are appreciated!

Running pimod with such an image currently gives the following error

```
mount: /tmp/tmp.qsBH33hJgz: wrong fs type, bad option, bad superblock on /dev/mapper/loop40p2, missing codepage or helper program, or other error.
### Error: "test.img" returned 32, cleaning up...
umount: bad usage

```

oxzi, 2021-09-23:
Thanks for creating this issue, as it shows a lack of documentation which I just closed via 622efc076f5800be0b71eb9d74dc21ff728f13d5.

By default, pimod assumes environment settings for a Raspberry Pi which uses the second partition for its root file system.
However, to support other targets with other partition tables, one can override this default with a second parameter for `FROM` or `INPLACE`. Support was introduced way back before the first release, but the documentation was missing.

Thus, I just started guessing partition numbers and just hit a good one with `1`.
The following example shows pimod running on the image you have provided in your release.

```
$ cat jetson.Pifile
INPLACE jetson.img 1
RUN echo hi

$ docker-compose run pimod pimod.sh jetson.Pifile
Creating pimod_pimod_run ... done
### FROM jetson.img 1
### TO jetson.img
Working inplace jetson.img.
add map loop5p1 (254:73): 0 5103616 linear 7:5 28672
[ . . . ]
add map loop5p14 (254:86): 0 256 linear 7:5 26624
update-binfmts: warning: qemu-arm already enabled in kernel.
update-binfmts: warning: qemu-armeb already enabled in kernel.
update-binfmts: warning: qemu-aarch64 already enabled in kernel.
### RUN echo hi
hi
umount: /tmp/tmp.8iAZYNLeou/dev/pts unmounted
umount: /tmp/tmp.8iAZYNLeou/dev unmounted
umount: /tmp/tmp.8iAZYNLeou/sys unmounted
umount: /tmp/tmp.8iAZYNLeou/proc unmounted
umount: /tmp/tmp.8iAZYNLeou unmounted
del devmap : loop5p9
[ . . . ]
del devmap : loop5p1
```

Please try it out and close this issue if everything looks good for you.

aniongithub, 2021-09-24:
I'll give it a shot and report back. AWESOME! :)

aniongithub, 2021-09-26:
Hmmm, the echo seems to work, but nothing else does. Here's one with just a PUMP that doesn't work. Of course, any subsequent commands fail with a "no space" message.

test.Pifile

```
FROM /home/ani/Downloads/jetson-nano-2gb/jetson.img 1

PUMP 100M
```

```
ani@jarvis01:~/Projects/pimod$ sudo ./pimod.sh test.Pifile
### FROM /home/ani/Downloads/jetson-nano-2gb/jetson.img
Copying /home/ani/Downloads/jetson-nano-2gb/jetson.img to test.img.
### PUMP 100M
1+0 records in
1+0 records out
104857600 bytes (105 MB, 100 MiB) copied, 0.0617087 s, 1.7 GB/s
GPT PMBR size mismatch (5138431 != 5343231) will be corrected by write.
The backup GPT table is not on the end of the device.

Welcome to fdisk (util-linux 2.36).
Changes will remain in memory only, until you decide to write them.
Be careful before using the write command.

GPT PMBR size mismatch (5138431 != 5343231) will be corrected by write.
The backup GPT table is not on the end of the device. This problem will be corrected by write.

Command (m for help): Partition number (1-14, default 14): 
Partition 1 has been deleted.

Command (m for help): Partition number (1,15-128, default 1): Value out of range.
Partition number (1,15-128, default 1): First sector (34-5343198, default 28672): Value out of range.
First sector (34-5343198, default 28672): Last sector, +/-sectors or +/-size{K,M,G,T,P} (28672-5343198, default 5343198): Last sector, +/-sectors or +/-size{K,M,G,T,P} (28672-5343198, default 5343198): 
Command (m for help): 
GPT:Primary header thinks Alt. header is not at the end of the disk.
GPT:Alternate GPT header not at the end of the disk.
GPT: Use GNU Parted to correct GPT errors.
add map loop50p1 (253:28): 0 5105664 linear 7:50 28672
add map loop50p2 (253:29): 0 256 linear 7:50 2048
add map loop50p3 (253:30): 0 896 linear 7:50 4096
add map loop50p4 (253:31): 0 1152 linear 7:50 6144
add map loop50p5 (253:32): 0 128 linear 7:50 8192
add map loop50p6 (253:33): 0 384 linear 7:50 10240
add map loop50p7 (253:34): 0 768 linear 7:50 12288
add map loop50p8 (253:35): 0 128 linear 7:50 14336
add map loop50p9 (253:36): 0 896 linear 7:50 16384
add map loop50p10 (253:37): 0 896 linear 7:50 18432
add map loop50p11 (253:38): 0 1536 linear 7:50 20480
add map loop50p12 (253:39): 0 128 linear 7:50 22528
add map loop50p13 (253:40): 0 384 linear 7:50 24576
add map loop50p14 (253:41): 0 256 linear 7:50 26624
/dev/mapper/loop50p1: 73637/159680 files (0.1% non-contiguous), 608651/638208 blocks
resize2fs 1.45.6 (20-Mar-2020)
The filesystem is already 638208 (4k) blocks long.  Nothing to do!

GPT PMBR size mismatch (5138431 != 5343231) will be corrected by write.
The backup GPT table is not on the end of the device.
Disk /dev/loop50: 2.55 GiB, 2735734784 bytes, 5343232 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: gpt
Disk identifier: 337BE1C0-9537-4692-8B58-D5A7EE140D14

Device         Start     End Sectors  Size Type
/dev/loop50p1  28672 5134335 5105664  2.4G Linux filesystem
/dev/loop50p2   2048    2303     256  128K Linux filesystem
/dev/loop50p3   4096    4991     896  448K Linux filesystem
/dev/loop50p4   6144    7295    1152  576K Linux filesystem
/dev/loop50p5   8192    8319     128   64K Linux filesystem
/dev/loop50p6  10240   10623     384  192K Linux filesystem
/dev/loop50p7  12288   13055     768  384K Linux filesystem
/dev/loop50p8  14336   14463     128   64K Linux filesystem
/dev/loop50p9  16384   17279     896  448K Linux filesystem
/dev/loop50p10 18432   19327     896  448K Linux filesystem
/dev/loop50p11 20480   22015    1536  768K Linux filesystem
/dev/loop50p12 22528   22655     128   64K Linux filesystem
/dev/loop50p13 24576   24959     384  192K Linux filesystem
/dev/loop50p14 26624   26879     256  128K Linux filesystem

Partition table entries are not in disk order.
del devmap : loop50p10
del devmap : loop50p4
del devmap : loop50p3
del devmap : loop50p2
del devmap : loop50p1
del devmap : loop50p9
del devmap : loop50p14
del devmap : loop50p8
del devmap : loop50p13
del devmap : loop50p7
del devmap : loop50p12
del devmap : loop50p6
del devmap : loop50p11
del devmap : loop50p5
GPT:Primary header thinks Alt. header is not at the end of the disk.
GPT:Alternate GPT header not at the end of the disk.
GPT: Use GNU Parted to correct GPT errors.
add map loop50p1 (253:28): 0 5105664 linear 7:50 28672
add map loop50p2 (253:29): 0 256 linear 7:50 2048
add map loop50p3 (253:30): 0 896 linear 7:50 4096
add map loop50p4 (253:31): 0 1152 linear 7:50 6144
add map loop50p5 (253:32): 0 128 linear 7:50 8192
add map loop50p6 (253:33): 0 384 linear 7:50 10240
add map loop50p7 (253:34): 0 768 linear 7:50 12288
add map loop50p8 (253:35): 0 128 linear 7:50 14336
add map loop50p9 (253:36): 0 896 linear 7:50 16384
add map loop50p10 (253:37): 0 896 linear 7:50 18432
add map loop50p11 (253:38): 0 1536 linear 7:50 20480
add map loop50p12 (253:39): 0 128 linear 7:50 22528
add map loop50p13 (253:40): 0 384 linear 7:50 24576
add map loop50p14 (253:41): 0 256 linear 7:50 26624
update-binfmts: warning: qemu-arm already enabled in kernel.
update-binfmts: warning: qemu-armeb already enabled in kernel.
update-binfmts: warning: qemu-aarch64 already enabled in kernel.
### PUMP 100M; file system usage:
Filesystem            Size  Used Avail Use% Mounted on
/dev/mapper/loop50p1  2.4G  2.3G     0 100% /tmp/tmp.IcnN0zyBkG
umount: /tmp/tmp.IcnN0zyBkG/dev/pts unmounted
umount: /tmp/tmp.IcnN0zyBkG/dev unmounted
umount: /tmp/tmp.IcnN0zyBkG/sys unmounted
umount: /tmp/tmp.IcnN0zyBkG/proc unmounted
umount: /tmp/tmp.IcnN0zyBkG unmounted
del devmap : loop50p10
del devmap : loop50p4
del devmap : loop50p3
del devmap : loop50p2
del devmap : loop50p1
del devmap : loop50p9
del devmap : loop50p14
del devmap : loop50p8
del devmap : loop50p13
del devmap : loop50p7
del devmap : loop50p12
del devmap : loop50p6
del devmap : loop50p11
del devmap : loop50p5

```

oxzi, 2021-10-02:
Thanks for your feedback, @aniongithub.
I have tried to trace down the issue, which is related to the huge amount of partitions. The changes in 5336a3092501ec258468672bfeb3de54d1d49f54 should fix this.
Please try out the latest code on the `master`'s `HEAD` and report back.

aniongithub, 2021-10-04:
Yay, it works! Thanks for the fix!

closed 2021-10-04 by aniongithub

oxzi, 2021-10-04:
Thanks for merging :-)