Nature40/pimod #37: Multiline RUN command not working on 0.4.0

adur1990, 2021-11-01:
In Pimod version 0.2.2, I used to use the following code:

```
RUN bash -c "
  cd somewhere
  do something here
  more commands here
"
```

But with Pimod 0.4.0, which was uploaded a few days ago, this fails with the following error:

```
### RUN bash -c
  cd somewhere
  do something here
  more commands here

/bin/bash: $n: command not found
### Error: "bash" returned 127, cleaning up...
```

jonashoechst, 2021-11-02:
I've created a small example to recreate the behavior, but was not able to reproduce the error:

```bash
FROM http://downloads.openwrt.org/releases/18.06.5/targets/brcm2708/bcm2710/openwrt-18.06.5-brcm2708-bcm2710-rpi-3-ext4-factory.img.gz

# test size of the image
RUN sh -c "
    echo hello
    echo world
"
```

jonashoechst, 2021-11-02:
`sh` of OpenWRT is kind of a special case and does not reproduce the problem clear enough. When using `bash -c` the error can be reproduces even in OpenWRT.

jonashoechst, 2021-11-02:
A central conceptual problem is that `printf ' %q'` does format the input as `bash`-escaped string. When supplying these strings to `sh`, misinterpretations can occur and seem to have happened here.

In the OpenWRT example `/bin/sh` supports the requested formatting options, but the Debian (vanilla) `/bin/sh` did not support the format string. 

A discussion of the probem can be found here: https://stackoverflow.com/questions/12162010/posix-sh-equivalent-for-bash-s-printf-q

A workaround through reformatting the `printf`'s output is implemented here: https://github.com/mentalisttraceur/esceval/blob/master/sh/esceval.sh

jonashoechst, 2021-11-02:
https://github.com/Nature40/pimod/blob/12e8ae4c62328bdb51c76739b0c85fda2162a602/stages/30-chroot.sh#L72-L73

Following the current implementation the guest OS requires `/bin/sh` to exist and to support the requested format. This additional shell could be circumvented, if there was a `chroot` option to specify a chdir, because `cd` inside this additional shell would not be required. 

As there is no such option, a custom `chroot` patch could be useful here.

closed 2021-11-04 by jonashoechst