Nature40/pimod #39: INSTALL does not copy hidden files

adur1990, 2021-11-03:
When using INSTALL, hidden files are not copied to the guest.
In my case, I try to build [nexmon](https://nemon.org), which in turn requires the .git folder to built.
See the output below.

```
### RUN bash -c
    cp -r /btp/nexmon_patch /nexmon/patches/bcm43430a1/7_45_41_46/btp/
    cd /nexmon/patches/bcm43430a1/7_45_41_46/btp/ || exit 1
    source /nexmon/setup_env.sh
    make

fatal: Not a git repository: ../.git/modules/nexmon
```

oxzi, 2021-11-09:
Thanks for reporting this bug, @adur1990. Could you please supply your Pifile and at least a directory listing of the `INSTALL`ed files resp. directories on your host?

The following is the current copy operation, depending on being a directory or a file.
https://github.com/Nature40/pimod/blob/1346b82b3e9c996100424e1e2f6e6dac0c3c60cf/stages/30-chroot.sh#L35-L39
In my understanding, a source directory should be copied to its destination with all its children. But, of course, I may be in the wrong here.

The used `-P` switch excludes symbolic links, as this might end up being messy. Might this be related to your issue?

sensslen, 2022-01-06:
Is there any reason why cp is used instead of the actual install command available?

oxzi, 2022-01-17:
Initially, the `INSTALL` command used the `install` program. This was changed in 8151607d8e7e61457108e9bb7e2d4cc3f099e181 for #5 and #8.

jonashoechst, 2022-10-13:
I tried to reproduce the error but was not able to. I've discussed with @adur1990 at it seems that the project structure he's using yielded the error. 

Hidden files inside of folders are copied using the current implementation of pimod, which was already pointed out by @oxzi.

closed 2022-10-13 by jonashoechst