Nature40/pimod #51: Can't create image on WSL2

aniongithub, 2022-01-16:
Running the included example with the latest commit from master gives me the following error during `umount`.

    (base) ani@ani-surface:~/pimod$ sudo ./pimod.sh examples/RPi-RaspberryPiOSLite.Pifile 
    ### FROM https://downloads.raspberrypi.org/raspios_lite_armhf/images/raspios_lite_armhf-2021-05-28/2021-05-07-raspios-buster-armhf-lite.zip
    Using cache: /var/cache/pimod/downloads.raspberrypi.org/raspios_lite_armhf/images/raspios_lite_armhf-2021-05-28/2021-05-07-raspios-buster-armhf-lite.zip

    7-Zip [64] 16.02 : Copyright (c) 1999-2016 Igor Pavlov : 2016-05-21
    p7zip Version 16.02 (locale=C.UTF-8,Utf16=on,HugeFiles=on,64 bits,8 CPUs Intel(R) Core(TM) i7-1065G7 CPU @ 1.30GHz (706E5),ASM,AES-NI)

    Scanning the drive for archives:
    1 file, 465662909 bytes (445 MiB)

    Extracting archive: /var/cache/pimod/downloads.raspberrypi.org/raspios_lite_armhf/images/raspios_lite_armhf-2021-05-28/2021-05-07-raspios-buster-armhf-lite.zip
    --
    Path = /var/cache/pimod/downloads.raspberrypi.org/raspios_lite_armhf/images/raspios_lite_armhf-2021-05-28/2021-05-07-raspios-buster-armhf-lite.zip
    Type = zip
    Physical Size = 465662909

    Everything is Ok

    Size:       1874853888
    Compressed: 465662909
    Moving temporary /tmp/tmp.B43wH7cX2S to examples/RPi-RaspberryPiOSLite.img
    ### PUMP 200M
    pump conversion to 1M * 200
    pump ceil: 1M * 200
    200+0 records in
    200+0 records out
    209715200 bytes (210 MB, 200 MiB) copied, 0.126136 s, 1.7 GB/s
    Checking that no-one is using this disk right now ... OK

    Disk examples/RPi-RaspberryPiOSLite.img: 1.96 GiB, 2084569088 bytes, 4071424 sectors
    Units: sectors of 1 * 512 = 512 bytes
    Sector size (logical/physical): 512 bytes / 512 bytes
    I/O size (minimum/optimal): 512 bytes / 512 bytes
    Disklabel type: dos
    Disk identifier: 0x9730496b

    Old situation:

    Device                              Boot  Start     End Sectors  Size Id Type
    examples/RPi-RaspberryPiOSLite.img1        8192  532479  524288  256M  c W95 FAT32 (LBA)
    examples/RPi-RaspberryPiOSLite.img2      532480 3661823 3129344  1.5G 83 Linux

    examples/RPi-RaspberryPiOSLite.img2: 
    New situation:
    Disklabel type: dos
    Disk identifier: 0x9730496b

    Device                              Boot  Start     End Sectors  Size Id Type
    examples/RPi-RaspberryPiOSLite.img1        8192  532479  524288  256M  c W95 FAT32 (LBA)
    examples/RPi-RaspberryPiOSLite.img2      532480 4071423 3538944  1.7G 83 Linux

    The partition table has been altered.
    Syncing disks.
    add map loop9p1 (252:14): 0 524288 linear 7:9 8192
    add map loop9p2 (252:15): 0 3538944 linear 7:9 532480
    rootfs: 44957/97920 files (0.4% non-contiguous), 295831/391168 blocks
    resize2fs 1.45.5 (07-Jan-2020)
    Resizing the filesystem on /dev/mapper/loop9p2 to 442368 (4k) blocks.
    The filesystem on /dev/mapper/loop9p2 is now 442368 (4k) blocks long.

    del devmap : loop9p2
    del devmap : loop9p1
    add map loop9p1 (252:14): 0 524288 linear 7:9 8192
    add map loop9p2 (252:15): 0 3538944 linear 7:9 532480
    umount: bad usage
    Try 'umount --help' for more information.
    umount: bad usage
    Try 'umount --help' for more information.
    umount: bad usage
    Try 'umount --help' for more information.
    umount: bad usage
    Try 'umount --help' for more information.

aniongithub, 2022-01-17:
Ignore this - I didn't have a few required packages on my WSL distro.

closed 2022-01-17 by aniongithub

oxzi, 2022-01-17:
Don't worry, thanks for reporting anyway. Please feel free to crate a Pull Request to explain how to use `pimod` within WSL in the README.