Nature40/pimod #53: Bulk execution not working

icecube45, 2022-01-18:
Not sure when this bug was introduced, as I last used Pimod on 0.2.2, but in 0.5, bulk execution is not working
e.g.
```
FROM https://downloads.raspberrypi.org/raspios_lite_armhf/images/raspios_lite_armhf-2021-11-08/2021-10-30-raspios-bullseye-armhf-lite.zip
PUMP 1000M

RUN <<EOF
touch test.txt
echo "Hello World" > test.txt
EOF

EXTRACT test.txt test.txt
```

results in
```
### RUN 
### EXTRACT test.txt test.txt
cp: cannot stat '/tmp/tmp.SIUScz53RS/test.txt': No such file or directory
### Error: "test.txt" returned 1, cleaning up...
```

oxzi, 2022-01-31:
Thanks for reporting this issue.

I am not quite sure when this change was introduced, but I would guess it happened when environment variables were introduced.

When executing code from an heredoc, it is piped to the shell's stdin.
However, after multiple substitutions were added, this "implicit" kind of execution is no longer supported.

In general, heredocs are still working, as I tested with the following snippet:

```
RUN tee -a test.txt <<EOF
foo
bar
EOF

RUN cat test.txt
```
```
### RUN tee -a test.txt
foo
bar
### RUN cat test.txt
foo
bar
```

Even when this might not be very satisfying, I would ask you to use
```
RUN sh -c '
touch test.txt
echo "Hello World" > test.txt
'
```
instead.
I have successfully tested your Pifile with this substituted `RUN` command.

closed 2022-01-31 by oxzi

icecube45, 2022-01-31:
@oxzi if this is the way to do bulk execution for now - I suggest updating the readme

oxzi, 2022-02-07:
I was not aware that we are still recommending this anywhere. Thus, I updated the README.

Thanks again for reporting this issue.