Nature40/pimod #54: Networking not working properly

sensslen, 2022-01-18:
In our build process we are using apt-cacher in order to capture all apt packages being installed. To do so, we spin up an apt cacher docker container (sameersbn/apt-cacher-ng:latest).

Intrestingly `HOST curl http://apt-cacher-ng:3142` works as expected where  `RUN curl http://apt-cacher-ng:3142` does not (of course I did make sure that curl ist installed both on the host and on the target.

How would I resolve this issue?

oxzi, 2022-02-07:
In 1386db68dd0bb6967d3d99944e4cbf3a5c27ab8b, I have introduced a new `--host-resolv` flag for pimod to always bind mount the host's `/etc/resolv.conf` file. This commit continued where #55 left off.

@sensslen: Could you please test and post your feedback. Thanks.

sensslen, 2022-02-08:
This does indeed resolve this concrete issue.

commit aba9e269 (2022-02-07): resolv_conf: flag for host's /etc/resolv.conf

Calling pimod with the newly introduced --host-resolv flag results in
the host's /etc/resolv.conf flag to be used over possibly existing
resolv.conf files within the guest's image.

This commit supersedes and closes #55. Thus, it also closes #54.
commit 1386db68 (2022-02-07): resolv_conf: flag for host's /etc/resolv.conf

Calling pimod with the newly introduced --host-resolv flag results in
the host's /etc/resolv.conf flag to be used over possibly existing
resolv.conf files within the guest's image.

This commit supersedes and closes #55. Thus, it also closes #54.