Nature40/pimod #57: Find clean way to edit resolv.conf if it was replaced/bind mounted

sensslen, 2022-02-08:
In my attempt to use pimod to build a customized raspberry pi image fro our appliance, I encountered teh followin issue:

- To provide our custom debian packages for installation I use  a docker container hosting an apt repository and then use docker networking capabilities to access it.
- As the default resolv.conf file does not allow to properly resolve the hostname by said socker container, I rely on the `--host-resolv` flag
- The modifications I make require that I replace openresolv with systemd-resolved. This requires me to change the resolv.conf file to link to `/run/systemd/resolve/stub-resolv.conf`.

Apparently the step described above is not possible in pimod (as resolv.conf is bind mounted to the hosts).
Unfortunately the current implementation of pimod makes it impossible to detect whether a script runs in pimod envidonment and also makes it impossible to change the original resolv.conf file, as it is moved out to the host.

I'm currently looking for a clean way to execute the described procedure. #55 did in fact allow me to be context aware and thus would have resolved my issue. On the other hand I totally understand that this is not a clean solution!

oxzi, 2022-02-08:
For me, those are two issues of their own, which can be addressed like the following.

First, introduce a second pimod flag to do the opposite of `--host-resolv`, never bind the host's `/etc/resolv.conf` to the client. This, however, might lead to errors due to the file's absence.
Nevertheless, if a modification to the `/etc/resolv.conf` file is made first, this should work, since the user is aware of this with the explicit flag.

Second, we could introduce such an environmental variable, as you have already suggested in #55.
Otherwise, as this is very specific to your use case, you could use the `ENV` command within your Pifile.

Please let me know if this would work for you.

sensslen, 2022-02-09:
Thinking about it, I guess you're right! thanks fpr the suggestion. This will work for me...

closed 2022-02-09 by sensslen