Nature40/pimod #59: Feat: A `SHRINK` command

she11sh0cked, 2022-05-10:
`SHRINK [SIZE]`

`SHRINK` decreases the size of the image to the minimum possible.
Optionally, you can specify the amount to decrease (suffixes K, M, G are allowed).

**Example:**

```docker
PUMP 10G # Increase the size of the image by 10G
[...] # Do some stuff
SHRINK # Decrease the size of the image to the minimum possible
```

oxzi, 2022-05-10:
Isn't there already [PiShrink](https://github.com/Drewsif/PiShrink) to do exactly this job? It is currently even linked in pimod's README.

I would see the benefit of including PiShrink into pimod as a `SHRINK` command which reduces the size to the minimum. However, I don't see any advantage of letting the user specify the size to be shrinked.

If you, or anyone else, wants to address this in a Pull Request, I'd be happy to merge it. Currently I don't have the time to do this by myself.

ethanjli, 2024-05-28:
It looks like this feature request is mostly a matter of making it more convenient to use pishrink's functionality in workflows involving Pimod. I'm building OS images for the Raspberry Pi exclusively in CI, and I found this feature request while exploring my options for running pishrink more easily on GitHub Actions. To simplify that use-case in a way that was easy for me to implement, I've now made a GitHub action ([ethanjli/pishrink-action](https://github.com/marketplace/actions/run-pishrink)) which is a slightly-nicer wrapper around pishrink.sh, e.g. as in this example for running pishrink on the output of Pimod: https://github.com/ethanjli/pishrink-action?tab=readme-ov-file#pimod-usage-example . 

(At least for now I'm not the right person to make a PR for integrating pishrink into pimod, because I'm currently not relying on Pimod for my projects: my projects need to build images which require a fully-booted OS because the setup process involves doing some stuff with the Docker daemon in the image, and it was simpler for me to just use a combination of `systemd-nspawn` for unbooted setup steps with a QEMU VM for booted setup steps. Regardless, I think Pimod is really cool, and if/when I do use Pimod for another project then I will think about implementing a `SHRINK` command!)

jonashoechst, 2025-12-19:
Based on the pull request by @aniongithub I've implemented the SHRINK command, which is available starting v0.9.0. https://github.com/Nature40/pimod/releases/tag/v0.9.0

closed 2025-12-19 by jonashoechst