Nature40/pimod #69: Is not generating the image anymore

vinicius73, 2023-07-16:
A fresh example is not working anymore.

```
### Warning: Remaining processes (1 5737 5739) are killed.
./modules/chroot.sh: line 35: kill: (5737) - No such process
./modules/chroot.sh: line 35: kill: (5739) - No such process
umount: /tmp/tmp.4QdRGKZiVv/dev/pts (devpts) unmounted
umount: /tmp/tmp.4QdRGKZiVv/dev (tmpfs) unmounted
umount: /tmp/tmp.4QdRGKZiVv/sys/fs/cgroup (cgroup) unmounted
umount: /tmp/tmp.4QdRGKZiVv/sys (sysfs) unmounted
umount: /tmp/tmp.4QdRGKZiVv/proc (proc) unmounted
umount: /tmp/tmp.4QdRGKZiVv/boot (/dev/mapper/loop0p1) unmounted
umount: /tmp/tmp.4QdRGKZiVv (/dev/mapper/loop0p2) unmounted
```

oxzi, 2023-07-16:
> ### Warning: Remaining processes (1 5737 5739) are killed.
> ./modules/chroot.sh: line 35: kill: (5737) - No such process
> ./modules/chroot.sh: line 35: kill: (5739) - No such process

Thanks for reaching out. Based on your output, one can only guess what
went wrong. The warning happens during the chroot_teardown function
when there are still processes left.

Could you please provide information about your system and the used
Pifile or a redacted version. And please tell us if this error is
reproducible?

oxzi, 2023-09-15:
As there were no further information, I am going to close this issue. Please feel free to reopen it with more details at any time!

closed 2023-09-15 by oxzi